vscode-arduino/vscode-arduino #8: ESP32 missing includes

Benik3, 2024-06-03:
Hi.
After longer time I opened project and updated Arduino and VSCode IntelliSense now report, that it's not able to find some basic include files from ESP IDF (e.g. FreeRTOS.h).
I found out, that IncludePath in _c_cpp_properties.json_ doesn't point to the include folder, only to **dio_qspi**.

E.g. this is included `"C:\\Users\\blabla\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083\\esp32\\dio_qspi\\include"`

but this not `"C:\\Users\\blabla\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083\\esp32\\include"`

I tried to add the path manually, but it's immediately rewritten by the generated includes.

Thanks

Johnson48858, 2024-06-28:
I can confirm that same issue with my Arduino Pontenta H7. After Disabling Intelli Sense Auto Gen and adding the paths, and there are a lot of them in my case. But after that everything went smoothly. I hope this gets fixed

Benik3, 2024-06-29:
I found out, that the problem is with the new ESP library 3.0  
With v2 it works still OK.

agualdron, 2024-10-09:
>  After Disabling Intelli Sense Auto Gen

Could you please let me know how to disable Intelli Sense Auto Gen in VSCode (Windows)?

Netexcavator, 2025-01-13:
I've tried disabling the auto-generation for the IntelliSense and adding in the correct includes, but it just ends up causing more missing includes with the Arduino.h. Has there been any additional progress with this issue?