vscode-arduino/vscode-arduino #13: Update and Maintain Community-Maintained Fork of Deprecated Arduino Extension

AppSolves, 2024-09-02:
## Description:

The official Microsoft VSCode extension for Arduino has been deprecated, leaving many developers and enthusiasts without a reliable and actively maintained tool. Thankfully, there is already a community-maintained fork of this extension that could fill the gap left by Microsoft's decision. However, this fork is currently 36 commits behind the original repository, meaning it is outdated and missing critical updates and improvements.

## Why This Matters:

The Arduino community relies heavily on tools like this extension to streamline and enhance development. With the official extension no longer being updated, it is vital that we step up and maintain the community fork to ensure it remains a robust and useful tool for all users. By updating the fork, we can continue to provide support, new features, and bug fixes that are essential for Arduino developers using VSCode.

## Call to Action:

We are reaching out to the community to contribute to this effort. Whether you can help by reviewing the commits, updating the codebase, testing the updated extension, or providing documentation - every contribution matters!

Here are some ways you can contribute:

    - Pull the Latest Changes: Review the 36 commits and merge relevant updates into the fork.
    - Test the Updated Extension: Help identify bugs and test new features to ensure the fork remains stable and reliable.
    - Enhance the Documentation: Make sure our documentation is up-to-date and provides clear instructions for installation and usage.
    - Spread the Word: Share this issue with others who might be interested in contributing.

Let’s work together to keep this essential tool alive and thriving for the Arduino community!

Thank you to everyone who has contributed so far, and we look forward to seeing this fork become even better with your help!

lersi, 2024-09-02:
Thank you for your concern, This repo looks like it is more commits behind than it really is, because I accepted some commits from Microsoft's repo and squashed committed them back to this repo (that's one of the 3 commits ahead).
I am going to add the latest commit from Microsoft today, and hopefully with time this repo will become the main and active one.

Sorry that I left this repo a little unmanaged, did not have much free time lately

lersi, 2024-09-02:
Just updated the branch with the latest of Microsoft, as you can see, the repo is still "36 commits behind" but is actually up to date with Microsoft

rkilchmn, 2024-09-03:
Happy to test and contribute. How do I install this repo in VSCode?

lersi, 2024-09-03:
Hello @rkilchmn glad that you want to help.
You can always download the artifacts from the latest GitHub action.
In addition, I uploaded the latest commit to vscode marketplace, under the name "Arduino Community Edition" so you can also just install the extension like any other.
I am going to mark the current commit as a release and upload the artifacts there for ease of access

VAleSh-Soft, 2024-09-03:
After the update, I get the following message

[image]

'View README' redirects to the page https://github.com/microsoft/vscode-arduino/blob/main/README.md

AppSolves, 2024-09-03:
@lersi @VAleSh-Soft I think the commit from yesterday not only synchronized this repo with Microsoft's codebase but also pulled all the commits that were intended to prepare for the deprecation of the original extension (including changes like the deprecation warning in `README.md`)

AppSolves, 2024-09-03:
The deprecation warning was accidentally merged too in `src/extension.ts`:

```
export function showDeprecatedPopup(): void {
    vscode.window.
        showWarningMessage(
            "The Arduino extension is deprecated. Please view our README for more information.",
            "View README",
        ).then((selection) => {
        if (selection === "View README") {
            vscode.commands.executeCommand("vscode.open", vscode.Uri.parse("https://aka.ms/arduinoextensionreadme"));
        }
    });
}
```

lersi, 2024-09-03:
@VAleSh-Soft You are right, I accidentally also synced their deprecation warnings oops... 
Thanks for letting me know

closed 2024-10-21 by lersi