vscode-arduino/vscode-arduino #24: Long term idea: move to clangd lsp

M0JXD, 2024-09-03:
Hello

First and foremost - thank you for this community edition of the Arduino extension.

I'm wondering if you might consider changing this extension to no longer depend on [Microsoft's C/C++ Extension](https://github.com/microsoft/vscode-cpptools)?

I ask this because I like to use VSCodium, but due to the Microsoft's extension having proprietary parts and licensing that does not allow it to be with open builds such as VSCodium. A popular alternative is the [clangd extension](https://github.com/clangd/vscode-clangd).

Of course when this was under Microsoft's jurisdiction that would never happen, but it is a possibility for the community edition!

Thank you.

lersi, 2024-09-03:
I will keep that in mind, but the current mindset is to keep things working without changing things too much, until there will be room for biger changes

per1234, 2024-09-03:
Just in case it will be relevant, I'll mention that [Arduino IDE 2.x](https://github.com/arduino/arduino-ide) uses clangd. This is done via a wrapper tool named "Arduino Language Server", which provides the necessary interpretation between the Arduino sketch framework and the standard C++ that clangd needs:

https://github.com/arduino/arduino-language-server

M0JXD, 2024-09-03:
Thanks, I didn't realise that Arduino was using its own customised clangd LSP in its latest editor, it'd probably be better to use that.

I can edit the issue's title to say something like "Long term idea: use the Arduino LSP" if so desired?

per1234, 2024-09-03:
> its own customised clangd LSP

To be clear, Arduino doesn't make any modifications to the clangd application itself. We run the stock clangd from LLVM (we do [build the clangd binary from source](https://github.com/arduino/clang-static-binaries) for ease of distribution, but without any modifications at all to the source code). But the separate "Arduino Language Server" tool sits between Arduino IDE and clangd.

M0JXD, 2024-09-04:
Oh right, sorry!

I noticed the README says this:

 "The Arduino Language Server is the tool that powers the auto-completion of the new [Arduino IDE 2](https://github.com/arduino/arduino-ide). It implements the standard [Language Server Protocol](https://microsoft.github.io/language-server-protocol/) so it can be used with other IDEs as well."
 
 I imagine that would still make using Arduino's wrapper the best choice? That way should Arduino team choose to make any customisations/extensions/whatever they would be easily brought to VS Code/ium.

rambrosteam, 2024-09-09:
@M0JXD  You're right. This is throwing up an error when I use with VScodium.
Is there any way to bypass this error, I just want the auto format feature
[image]

@lersi thank you for keeping this extension alive

cardcraft, 2024-09-29:
I was going to come here to request the same thing, to keep it as foss as possible and compatible with codium. It's worth considering as an option even if it would be in beta.

navr32, 2025-09-17:
Hi all!  congratulation for all this work. 

So this will very good if we are able to use it without too many tweak on vscodium and better to use the same clangd than arduino ide v2. And this will be good to have the extension publish on open-vsx.org . But i know many works to do..if someone have some free time..

So for waiting i am abble  On linux (but must worked on win too if found the good path where put the product.json) i have found a tweak to be abble to use on   VSCODIUM  the "arduino community edition"... 

So add a file name :  product.json

in /home/myhome/.config/VSCodium  

file content..

```
 {
  "nameShort": "Visual Studio Code",
  "nameLong": "Visual Studio Code",
  "extensionsGallery": {
    "serviceUrl": "https://marketplace.visualstudio.com/_apis/public/gallery",
    "cacheUrl": "https://vscode.blob.core.windows.net/gallery/index",
    "itemUrl": "https://marketplace.visualstudio.com/items"
  }
}

```
So now you have on the plugin search in vscodium that display the ms marketplace name plugin so you can search after arduino community edition and found it and click install...the dependencies off missing module are installed automatically..so no need to search after ms-code-cpp..and serial terminal.. vsix or have to generate and manual install them restart and open an arduino directory project and the plugin work..but you have to config and select the cli-arduino  checkbox in the arduino community edition settings... so after click on build at top it ask to select a board..so select at the bottom right your board, usb port , prog mode...and after this build and this build ok... .But for now i have bad highlight syntax..i have try cpp ..arduino output..re generate intellisense but always bad syntax higlight...If someone found the solution.hopping this helps . Thanks