vscode-arduino/vscode-arduino #26: Linux: arduino-cli.app disappeared in version 0.7.0

SpintroniK, 2024-09-04:
Upgrading to version 0.7.0 on debian today caused an error message on startup:
```text
The bundled Arduino CLI could not be marked executable. Ensure that VS Code is able to execute: /home/jeremy/.vscode/extensions/vscode-arduino.vscode-arduino-community-0.7.0-linux-x64/assets/platform/linux-x64/arduino-cli/arduino-cli.app
```

The `arduino-cli.app` doesn't exist any more, as shown here:
```bash
ls -lh /home/jeremy/.vscode/extensions/vscode-arduino.vscode-arduino-community-0.7.0-linux-x64/assets/platform/
total 4.0K
drwxrwxr-x 3 jeremy jeremy 4.0K Sep  2 20:37 darwin-arm64
```

arvitus, 2024-09-05:
same for windows. It seems like it doesn't get installed automatically. When downloading the cli manually and placing it in the intended folder, everything works though.

vicatcu, 2024-09-05:
Confirmed downloading the latest release from https://arduino.github.io/arduino-cli/0.35/installation/, extracting the executable, creating the requisite folder structure in the extensions, moving / renaming the executable to what the extension is expecting (arduino-cli.app), then `chmod +x`ing the executable seems to work.

lersi, 2024-09-08:
Please let me know if this issue persists in release 0.7.1

spusuf, 2024-09-09:
> Please let me know if this issue persists in release 0.7.1

The issue does persist yes, there is a darwin-arm64 but no linux-x64 folder as expected in the error message. Making the directory and downloading arduino cli and setting as executable manually does not work. Same error is provided

serek4, 2024-09-10:
this works for me
> Confirmed downloading the latest release from https://arduino.github.io/arduino-cli/0.35/installation/, extracting the executable, creating the requisite folder structure in the extensions, moving / [...]

btw latest version is [1.0.4](https://arduino.github.io/arduino-cli/1.0/)

lersi, 2024-09-10:
Does this also happens if you install the extension from VSIX file (available on the release page)?
The issue seems to happen only when installing from market place

vicatcu, 2024-09-10:
@lersi Sorry I can't help confirm that, I've only ever installed extensions through the marketplace-- I don't know how from the command line.

lersi, 2024-09-10:
From farther digging the extension files in the market place are messed up, and the serve the wrong VSIX file for the architecture.
I need to make sure if I did something wrong or is there a problem with their publish tool

lersi, 2024-09-10:
Their tool is so messed up, I will now run through versions until I find the correct way of publishing

lersi, 2024-09-10:
Please let me know if this issue persists in version "0.7.2" (dummy version)

lersi, 2024-09-11:
So the problem was indeed with vsce tool, closing this issue as solve

closed 2024-09-11 by lersi