vscode-arduino/vscode-arduino #27: Win bundle arduino-cli not found

smoluks, 2024-09-07:
Steps to reproduce:

1. Install plugin (v0.7.0)
useArduinoCli set up to true by default:
arduino.path and arduino.commandPath clear by default

2. See console:
[Error] Analyzing sketch 'firmware.ino': {"error":{"errno":-4058,"code":"ENOENT","syscall":"spawn c:\\Users\\Administrator\\.vscode\\extensions\\vscode-arduino.vscode-arduino-community-0.7.0-win32-x64\\assets\\platform\\win32-x64\\arduino-cli\\arduino-cli.exe","path":"c:\\Users\\Administrator\\.vscode\\extensions\\vscode-arduino.vscode-arduino-community-0.7.0-win32-x64\\assets\\platform\\win32-x64\\arduino-cli\\arduino-cli.exe","spawnargs":["compile","-b","esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=app3M_fat9M_16MB,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=16M,UploadSpeed=921600,LoopCore=0,EventsCore=0,DebugLevel=info,EraseFlash=none,ZigbeeMode=default","--verbose","--no-color","--build-path","c:\\Users\\Administrator\\Desktop\\clock_esp\\firmware\\build","c:\\Users\\Administrator\\Desktop\\clock_esp\\firmware\\firmware.ino","--additional-urls",""]}}

Win files are missing in the directory, only darwin:
[image]

lersi, 2024-09-08:
Okay, managed to narrow down the problem, it was a compilation mistake that included the wrong set of tools, sorry I will fix it ASAP

lersi, 2024-09-08:
Please let me know if this issue persists in release 0.7.1

smoluks, 2024-09-08:
Yep
[image]
[image]

vicatcu, 2024-09-09:
Yes, still a problem. Had to re-do the fix when something automatically updated today when I opened VSCode.

vicatcu, 2024-09-09:
When I try and open from the command palette: "Arduino: Board Config" I get an error now (v0.7.1):
```
Error loading webview: Error: Could not register service worker: InvalidStateError: Failed to register a ServiceWorker: The document is in an invalid state..
```

lersi, 2024-09-10:
Please let me know if this issue persists in version "0.7.2" (dummy version)

vicatcu, 2024-09-10:
0.7.2 update didn't cause any trouble. Works as though I had manually patched the previous version.

lersi, 2024-09-11:
This problem was caused by vsce tool, closing this issue as solve
Thanks for the info and the help