vscode-arduino/vscode-arduino #42: Does it work with the current released versions of Arduino-cli like 1.0.4?

KurtE, 2024-10-16:
I have tried running it, by checking the box to use the CLI and 
setting the address of where I expanded the executable into from the .tar.gz file

Note: I am trying it running on a Raspberry PI 5 64 bits, so I downloaded arduino-cli_1.0.4_linux_ARM64.tar.gz
I am able to build and upload sketches to the board I am playing with, which is a Teensy MicroMod.  
This all works from a command prompt.

But if I insert the address to this CLI, most everything within Visual Studio Code stops working.  Things like, trying to bring
up the board manager, or select a board, upload...

However if I remove the path from the setting: Arduino Path:
Then restart visual code, it downloads all of the indexes, and I am able to choose the Teensy, plus choose the port and settings and do a build and upload operations.

Thanks for everyone who supports this!
Kurt

adiazulay, 2024-10-28:
Are you setting the Arduino Path to the CLI executable or the directory containing it? Based on the [readme](https://github.com/vscode-arduino/vscode-arduino?tab=readme-ov-file#arduino-cli) it should be the directory.

I'm going off of the readme since I haven't contributed to the code in a couple of years.

KurtE, 2024-10-30:
Thanks @adiazulay .

I have tried setting the arduino path to: /home/kurte/bin
I have also tried: /home/kurte/bin/arduino-cli

Likewise I have tried these with command path, left blank, and these two other options, and so none of them work.
My bin directory looks like:
```
kurte@RPI5:~/bin $ pwd
/home/kurte/bin
kurte@RPI5:~/bin $ ls -l
total 30672
-rwxr-xr-x 1 kurte kurte 31396307 Oct 10 15:46 arduino-cli
drwx------ 2 kurte kurte     4096 Oct 14 11:37 arduino-cli_1.0.4_Linux_ARM64
-rw-r--r-- 1 kurte kurte      113 Oct 11 07:05 arduino-cli.yaml
kurte@RPI5:~/bin $ 
```

And should mention, that this is on my path and the cli runs.
```
kurte@RPI5:~/bin $ arduino-cli version
arduino-cli  Version: 1.0.4 Commit: a0d912da Date: 2024-08-12T13:42:39Z

```