vscode-arduino/vscode-arduino #43: Unable to perform OTA updates

billy27607, 2024-10-28:
When one tries to perform an OTA update it fails.

When attempting using the Arduino upload command the error message "user input not supported in non interactive mode" is thrown. This appears to be because a password is expected and there is no mechanism to enter it. This happens even if the password requirement is disabled. 

When attempting using the Arduino CLI upload  command the error message "Error: unknown flag: --build-path" is thrown. There is no --build-path flag in the current arduino-cli upload command.

Between the two this effectively breaks the use of ota in projects. This is reproducible.  Is there a way to specify a password to arduino-cli to make this work?

lersi, 2024-10-28:
Does it not work also if you make the extension use locally installed Arduino CLI?
I want to make sure if there is a problem with the bundled Arduino CLI tools.

adiazulay, 2024-10-28:
This might be related: https://github.com/esp8266/Arduino/issues/8962

billy27607, 2024-10-28:
I can try that.  Where do I specify the path? I have the brew installed version. Billy

lersi, 2024-10-28:
In vscode setting, in the Extension's settings, edit: "Arduino command path"

[image]

billy27607, 2024-10-28:
I am working with esp32c3 devices.  I can successfully update its using the ide. Makes me think is something in the extension. Billy

billy27607, 2024-10-28:
With the brew installed arduino-cli I get a different error. In this case I get

“A fatal error occurred: Could not open 192.168.101.132, the port doesn't exist
Error during Upload: Failed uploading: uploading error: exit status 2"

The port does exist and I can upload to it using the IDE.

Billy

image.png (view on web)<https://github.com/user-attachments/assets/bbf6fd53-6bde-4ce0-8c3c-4c6b32fd25c9>

per1234, 2024-10-28:
> Is there a way to specify a password to arduino-cli to make this work?

Yes. Use the `--upload-field` flag:

https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_upload/#options

>```text
> -F, --upload-field key=value        Set a value for a field required to upload.
>```

For example:

```text
arduino-cli upload --upload-field "password=correcthorsebatterystaple" ...
```

In case you want to learn about how these "user provided fields" are configured in Arduino boards platforms, the documentation is here:

https://arduino.github.io/arduino-cli/latest/platform-specification/#user-provided-fields

You can see it done by the "**esp32**" boards platform here:

https://github.com/espressif/arduino-esp32/blob/3.0.7/platform.txt#L330-L332

billy27607, 2024-10-29:
Thank you, this is useful. If I am manually entering the arduino-cli commands I can uses the flags to get the right thing, The issue I need to solve is a way to pass the upload fields through the Arduino vscode extension to the arduino-cli. I asm not that familiar with eh construction of the extension.

What would be great would be to have a filed in the arduino.json file to specify auth for ota.

Billy

https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_upload/#options

https://arduino.github.io/arduino-cli/latest/platform-specification/#user-provided-fields

https://github.com/espressif/arduino-esp32/blob/3.0.7/platform.txt#L330-L332

billy27607, 2024-10-30:
Forgive me, but I am not really familiar with vscode extensions. If there a way to have this extension show the arduino-cli commands that it is issuing for the compile and build steps.