vscode-arduino/vscode-arduino #45: CVE-2024-43488: RCE vulnerability in original vscode-arduino code

ahmsec, 2024-10-31:
Hello folks,

The original https://github.com/microsoft/vscode-arduino code (which this repo forked from) has a remote code execution vulnerability. That is why the extension was deprecated. Please see the link below for details. 

https://msrc.microsoft.com/update-guide/vulnerability/CVE-2024-43488

It may be worth validating whether the vulnerability was transferred to this repo.

lersi, 2024-10-31:
Hello Ahmad,
Thanks for letting me know.
This fork is almost fully 1:1 with microsoft’s repo, the difference is only minor bug fixes and improvements.

I cannot see technical details about the vulnerability on microsoft’s page, do you know where can I find additional details so I could fix this ASAP.

ahmsec, 2024-11-01:
@MatthewSteeves Fair enough. When I reported this vulnerability to Microsoft, they said their resolution was to deprecate the plugin. So the vulnerability may have been a contributing factor, if not the entire reason, for deprecation.

@lersi The core issue is that the plugin launches an unauthenticated localhost webserver. This webserver exposes sensitive capabilities. Anyone who can send network requests to this webserver (say a malicious webpage) can exploit the capabilities of the webserver.

To reproduce:
1. Activate the plugin
2. Run `sudo netstat -tunpl` and note the webserver's port
3. Run `curl -X POST -H 'Content-Type: application/json' --data '{"link": "https://www.example.com/"}' http://localhost:55842/api/openlink` (but first update the port).

The full RCE exploit is a bit more involved. I plan to share it at a later time. But the crux of the matter is this exposed webserver.

My suggestion would be to remove the webserver entirely, and use a different IPC mechanism between the UI and the code that invokes arduino-cli.

MatthewSteeves, 2024-11-02:
@ahmsec  Ahh ok - interesting! Given their response, I would agree it may well have factored in. Thanks for the additional background.

ahmsec, 2025-11-30:
Further details shared here: https://www.ahmsec.io/blog/2025/11/30/rce-in-vscode-arduino-extension/

closed 2026-10-25 by lersi