vscode-arduino/vscode-arduino #50: request: option to set the package / libraries path in settings

Henry-Jones, 2024-11-03:
The change **"Added ARDUINO_DIRECTORIES_DATA env variable to define package directory"** is very helpful.

I would suggest to take this approach a step further to have both VSCode and the arduino plugin portable.

As it currently is, either the arduino path contains "portable" or the mentioned env variable is set, to change the package path (arduinosettings.ts). The first one is unlikely, if the bundled CLI version is used. The second works but it must be set before starting VSCode which is cumbersome, at least until the possibility to set env variables during VSCode startup arrives (if ever) :)

So if there would be an option "Package Path" and "Library Path" to set those paths and be ignored when empty, this would be a great addition. I'm not that fluent in TypeScript to come up with a pull request (at least not yet).

For the beginning it would be enough to set them to fixed paths, as the removable drive has the same drive letter on all my development systems. As an example for the Python environment, I use the following setting:

    "python.defaultInterpreterPath": "${execPath}/../../CommonFiles/Python/python.exe",

This even allows to set the path relative to the VSCode executable file,

chinasmooth, 2024-12-04:
I agree; encountering errors while using CLI header files can be incredibly frustrating. 
option to set the package / libraries path in settings

goodchip, 2025-01-10:
> I agree; encountering errors while using CLI header files can be incredibly frustrating. option to set the package / libraries path in settings

I created a pull request (#59) with an option in the configuration to have Arduino-CLI read its configuration from an arduino-cli.yaml file placed in the root of the project. This provides access to almost all Arduino-CLI customizations, including the paths to your libraries.