vscode-arduino/vscode-arduino #52: Unexpected problem-causing definition of initVariant()

Sergio-Slobodrian, 2024-11-05:
When attempting to use initVariant in a library I'm creating for the Arduino Nano Every I was getting linker errors about multiple definitions of the initVariant function which is weakly defined in main.cpp specifically so it can be defined elsewhere.

The issue is the following. At the bottom of the file: .arduino15/packages/arduino/hardware/megaavr/1.8.8/variants/nona4809/variant.c an apparently random definition of the initVariant() function exists. It's empty and it's the last function in the file with no comments as to why it's there. 

The file came with the download when the board was defined. 

The board was defined as follows:
Arduino megaAVR Boards by Arduino Version 1.8.8 INSTALLED
Boards included in this package:
Arduino UNO WiFi Rev2, Arduino Nano Every

I selected the 4809 version.

The fix is simple, remove the function from the file. It shouldn't be there as an empty shell anyway and prevents the initVariant function from being used for its intended purpose. I just commented it out in the file for now but it would be nice to not have to go find and comment it out for each future upgrade.

per1234, 2024-11-05:
Hi @Sergio-Slobodrian. That `variant.c` file is not part of this repository's codebase so this is not the appropriate place to report the issue.

Sergio-Slobodrian, 2024-11-05:
Ok, I'll try to find out what repository it belongs to and report it there. I'll close this issue.