vscode-arduino/vscode-arduino #53: Please update bundled Arduino CLI version to 1.1.0

kiwiant, 2024-11-17:
For some reason after various updates this weekend, I was unable to upload sketches from Visual Studio Code (vscode) to my Arduino UNO R4 Wi-Fi with the error:

```No device found on COM3```

Both the Arduino IDE 2.3.3 (includes arduino-cli 1.0.4) and Arduino CLI standalone 1.1.0 had no issues uploading sketches.

The solution was to replace the copy of **arduino-cli** included in the extension (0.31.0) with the latest standalone copy (1.1.0).

```C:\Users\<username>\.vscode\extensions\vscode-arduino.vscode-arduino-community-0.7.2-win32-x64\assets\platform\win32-x64\arduino-cli\arduino-cli.exe"```

Running the included CLI version 0.31.0 gives the following message:
```
A new release of Arduino CLI is available: 0.31.0 → 1.1.0
https://arduino.github.io/arduino-cli/latest/installation/#latest-packages

```

tshcherban, 2024-12-08:
@lersi is that fork alive?

joaorr3, 2024-12-17:
Does your solution includes specifying a custom `Arduino: Path` and `Arduino: Command Path`?