vscode-arduino/vscode-arduino #54: Bundled version of Arduino CLI doesn't run on mac os

adiazulay, 2024-11-19:
I haven't been able to to get the extension to work on mac with the bundled Arduino CLI. I suspect the issue has to do with some Mac security setting that blocks software from "unknown" developers. I downloaded the CLI and pointed the extension to it and then had to go into my system settings to allow the CLI to run.

Here is the terminal output I get with the bundled CLI:
`[Starting] Verifying sketch 'arduino_ext.ino'
[Warning] Output path is not specified. Unable to reuse previously compiled files. Build will be slower. See README.
[Warning] Failed to generate IntelliSense configuration.
[Error] Verifying sketch 'arduino_ext.ino': {"code":null}`

joaorr3, 2024-12-17:
Just to clarify, did you get it working after specifying a custom `Arduino: Path` and `Arduino: Command Path`?

gandarez, 2024-12-18:
I'm using the bundled version with no issues. Make sure you set `"arduino.useArduinoCli": true` and **do not set** `arduino.path` and `arduino.commandPath`.

gandarez, 2024-12-18:
Oh I see, also make sure you disable testing the sketch with `"arduino.disableTestingOpen": true`.

joaorr3, 2024-12-27:
> @joaorr3 it works if I specify `Arduino: Path` @gandarez With that configuration I get the error I posted above.
> 
> I started working on updating the bundled version of the CLI since the version included is very old, but I haven't been able to get it to build on my computer yet.

Yeah, same here. After specifying the `arduino.path` (in my case the brew installation) everything went smoothly.

The option [`arduino.path`](https://github.com/vscode-arduino/vscode-arduino/blob/bbb99eb8620b2609cb0fcd55d589ee9610340140/README.md?plain=1#L85) is very misleading though. It suggests _"`/Applications` for Mac"_ which makes no sense at all. The option should be called something like: `arduino.cli-path` or `arduino.executable-path` and instead of suggesting a `/Applications` it should ask for the user to insert the output of `which arduino-cli` for ex.

Thanks for the help @adiazulay!

[image]

joaorr3, 2024-12-27:
Also, if anyone is getting trouble with intelliSense, make sure you specify the correct `output` path. 
In the `.vscode/arduino.json`, vscode may trick you into doing something like `"output": "../build"`, don't do this, instead, specify a path as if you're already at the root of the sketch folder, like: `"output": "build"`.

ckarcz, 2025-03-30:
I got VSCode development environment mostly working on my mac thanks to this thread 🎉

👉  my current issue is I get intellisense warnings/errors like
```identifier "TIMSK0" is undefined```
though when you click on the reference it opens a modal that shows hundreds of possibe definitions for the identifier. so my [intellisnese] environment is not tied to my specific board or library version or something.
example path to one definition of `TIMSK0`:
```
/Users/ckarcz/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include/avr/iocanxx.h
```
 
⭐  here is my setup that is working - verifiying code and deploying code to my nano:
1. Have Arduino IDE installed.
2. Install `arduino-cli` using hombrew.
3. Install the VSCode Arduino Community extension.
4. Edit your VSCode user or workspace settings to:
  - Set "Arduino: Use Arduino Cli" to true.
  - Set "Arduinio: Path" to `/opt/homebrew/bin/arduino-cli`.
5. Delete any `c_cpp_properties.json` file created before, and reload VSCode and open an "ino" file.
6. The extensiuon should generate a `c_cpp_properties.json` file that includes paths to the standard Arduino libraries.

here is an example of my cpp props file
```
{
    "version": 4,
    "configurations": [
        {
            "name": "Arduino",
            "compilerPath": "/Users/ckarcz/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++",
            "compilerArgs": [
                "-w",
                "-std=gnu++11",
                "-fpermissive",
                "-fno-exceptions",
                "-ffunction-sections",
                "-fdata-sections",
                "-fno-threadsafe-statics",
                "-Wno-error=narrowing"
            ],
            "intelliSenseMode": "gcc-x64",
            "includePath": [
                "/Users/ckarcz/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino",
                "/Users/ckarcz/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/variants/eightanaloginputs",
                "/Users/ckarcz/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/lib/gcc/avr/7.3.0/include",
                "/Users/ckarcz/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/lib/gcc/avr/7.3.0/include-fixed",
                "/Users/ckarcz/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include"
            ],
            "forcedInclude": [
                "/Users/ckarcz/Library/Arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/Arduino.h"
            ],
            "cStandard": "c11",
            "cppStandard": "c++11",
            "defines": [
               // these i left out for brevity
            ]
        }
    ]
}
```