vscode-arduino/vscode-arduino #62: Problem with missing #include <Arduino.h> while "arduino.useArduinoCli": true

ceewanna, 2025-01-19:
Using bundled Arduino Cli always give this red error "_#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit_" while it can manage to build. This happens to a simple file such as "Arduino.h".

If Arduino IDE is used, there is no problem/error.

OffApparition, 2025-02-08:
I have the same issue. Per default these are my included paths

```
"includePath": [
                "/home/off/Documents/Programming/Arduino/test",
                "/home/off/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-cfea4f7c-v1/esp32c3/qio_qspi/include",
                "/home/off/.arduino15/packages/esp32/hardware/esp32/3.1.1/cores/esp32",
                "/home/off/.arduino15/packages/esp32/hardware/esp32/3.1.1/variants/nologo_esp32c3_super_mini",
                "/home/off/.arduino15/packages/esp32/tools/esp-rv32/2405/riscv32-esp-elf/include/c++/13.2.0",
                "/home/off/.arduino15/packages/esp32/tools/esp-rv32/2405/riscv32-esp-elf/include/c++/13.2.0/riscv32-esp-elf",
                "/home/off/.arduino15/packages/esp32/tools/esp-rv32/2405/riscv32-esp-elf/include/c++/13.2.0/backward",
                "/home/off/.arduino15/packages/esp32/tools/esp-rv32/2405/lib/gcc/riscv32-esp-elf/13.2.0/include",
                "/home/off/.arduino15/packages/esp32/tools/esp-rv32/2405/lib/gcc/riscv32-esp-elf/13.2.0/include-fixed",
                "/home/off/.arduino15/packages/esp32/tools/esp-rv32/2405/riscv32-esp-elf/include"
            ],
"forcedInclude": [
                "/home/off/.arduino15/packages/esp32/hardware/esp32/3.1.1/cores/esp32/Arduino.h"
            ],
```

I've noticed that the Arduino.h file contains a lot of other header files which can not be found.
I can compile and upload the code without any problems but vscode is still highlighting a lot of the code in red and autocompletion doesn't work properly.
Here's a screenshot 

[image]

I'm on ArchLinux using VSCodium and an ESP32C3.

OffApparition, 2025-02-09:
There seems to be an issue with the esp32 boards version 3.0.0 and above. When rolling back to version 2.0.17 I was able to get it working. At least for me this temporary fix works but there are some esp32 boards like the esp32c6 which are not supported until later versions.
In [this](https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#apis) entry of the esp32 docs by espressif they list some breaking api changes from versions 2 to 3. Not too sure how or if those changes affect this vscode extension but rolling back to 2.0.17 definitely unbreaks it.

iadminko, 2025-02-19:
> There seems to be an issue with the esp32 boards version 3.0.0 and above. When rolling back to version 2.0.17 I was able to get it working.

Saved my day!
Thanks!